abersailbot/captain-h-morgan #4: Alternative methods of Arduino programing

naturesyouth, 2016-12-02:
_From @naturesyouth on February 14, 2016 14:54_

It would be nice if someone could look into the possibility of programming the Arduino without a usb connection, either over spi or serial, preferably with some hardware experimentation. This is in refrence to #45 

_Copied from original issue: abersailbot/dewi-hardware#48_

naturesyouth, 2016-12-02:
_From @Danyc0 on February 14, 2016 15:0_

So this page explains how to program an arduino using another arduino:
https://www.arduino.cc/en/Tutorial/ArduinoISP
I reckon if you reimplemented the ArduinoISP example (found built into the IDE these days) for the Pi it might just be possible. However I think that would take too much time to be worth it just for the purpose of getting rid of a USB cable

naturesyouth, 2017-10-19:
Becuase we should barely ever have to reprogram the embedded controller the programming pins can just be broken out to a header and then we can just program it with an external programmer.

Danyc0, 2017-10-22:
Being able to reprogram the arduino from the Pi without having to open any boats or boxes was an explicit design goal. It's super useful, especially in Kitty where getting the boxes out isn't the easiest thing in the world. It's also super useful for remote testing, where the boat is connected to the internet and everything is reprogrammed remotely