abersailbot/captain-h-morgan #7: Test using the MAX14830 for serial multiplexing.

naturesyouth, 2017-06-16:
I would like like to try using the MAX14830 to add some extra UART ports to the Pi, there should be a linux driver.

Danyc0, 2017-06-16:
Am I correct in reading that this is an SPI/I2C to UART bridge?

naturesyouth, 2017-06-16:
yeah, spi to 4 uart

naturesyouth, 2017-06-16:
for a total of five uarts on the pi.

rpasc, 2017-06-29:
Hi

MAX14830 is an option.

For the SPI interface there exists a module in the mainstream kernel http://elixir.free-electrons.com/linux/v4.12-rc7/source/drivers/tty/serial/max310x.c

I have shared a version that works with the I2C interface:
https://groups.google.com/d/msg/acmesystems/xLCdHYX81HA/nFw6HgwUOQAJ

This requires working at the kernel level and device trees. You will have to compile the kernel.

Another option, which does not require kernel level knowledge, is to use a USB to UART HUBs based on for example FTDI  FT4232H http://www.ftdichip.com/Products/Modules/DevelopmentModules.htm#FT4232H_Mini

This latter is around €27.

The boards I have with 14830 were custom designed and required soldering in an SMD oven... will be more expensive for small batches. I did get an Evaluation Kit at first, over €100...

Danyc0, 2017-06-29:
Sadly we have no free USB ports. We had a USB hub in Kitty and used an FTDI, but it just took up loads of space, something Captain Morgan is pretty much designed to try and save

naturesyouth, 2017-06-29:
Yeah part of the whole long term goal is to remove the need for USB devices and it's not out of the realm of possibility to solder a qfn by hand with the right footprint. I would be interested in knowing what sort of baudrate speeds you were able to get reliably with the i2c bus.

rpasc, 2017-06-30:
ah! only now I noticed this is for a robotic sailboat. Marine engineer on this side.

I was interested in getting 8 UART for an automation project, did so by using two 14830 on I2C. However, the board I used was the ARIAG25 and the module was the one I shared for kernel level. I did not test in RPi. I did not require high rate, 9600 bps is what I've been using on all ports at once. The device works in I2C but has buffer and interrupt driven reading.

I think you could get away soldering with hot air. Do you use Eagle PCB design software? I have schematics of the working 8 port circuit and would reduce it to 4 and share if that could be of use.

naturesyouth, 2017-06-30:
We use the KiCAD EDA package because it is open source, but I would love to see your schematic for reference, I hadn't considered using the i2c bus as I assumed that it would be too slow, but if you got it running at 9600 then it might be an interesting solution.

rpasc, 2017-07-03:
Should be enough to evaluate if you want to go that way.
[I2CUART.zip](https://github.com/abersailbot/captain-h-morgan/files/1120240/I2CUART.zip)

sbharsha, 2017-07-12:
I am using raspberry pi. can you post the device tree for MAX14830 over SPI. I am struggling to get the MAX14830 get work.

rpasc, 2017-07-12:
Sorry but I haven't used SPI or RPi with the device.

If you have an oscilloscope I would recommend that you try to put hardware problems aside, otherwise, if you have not tried I2C, I would recommend some common ground with respect to what I have. This means I2C, which should be relatively simple in this case because SPI and I2C pins are shared on the device. Enable I2C on the RPi.

Then with everything connected I would use shell command $i2cdetect -y dev

dev depends on which I2C bus you are using (0 or 1).

Only once hardware problems are set aside and the device is detected (try also without device to make sure it is the device that is responding) then you could try the driver I modified and the part of device tree which respects to the device.
https://groups.google.com/d/msg/acmesystems/xLCdHYX81HA/nFw6HgwUOQAJ

For I2C device at 6C

	max14830: max14830@6c {
                	compatible = "maxim,max14830";
                	reg = <0x6C>; // Just address of UART0, others will be inferred        
                	maxim,clock-type = "xtal";
                	maxim,clock-frequency = <3686400>;
               		interrupts-extended = <&pioC 1 2>; //IRQ_TYPE_EDGE_FALLING
               		maxim,minorstart = <209>; //209 name will be ttyMAX0
               	};

Droid-MAX, 2018-10-14:
@sbharsha me too!

jaredvis, 2021-08-04:
Can I ask if @Droid-MAX or @sbharsha managed to get the MAX14830 working via SPI, I would really appreciate any assistance on this one.